MangoIV/dracuLad #6: Pimoroni firmware

hyphonetics, 2021-08-10:
So I built the keyboard and it works except for the trackball that I have on the right side of the board. When I plug in only the right half, the pimoroni works fine, but when I link that to the left half and use the left as the master side it doesn't function.
I saw in the keymap something about commenting out part of the code on the side without the trackball but I'm not sure what exactly I need to do to get it functioning properly, any help would be greatly appreciated.

MangoIV, 2021-08-10:
Hi! Using the pimoroni trackball on the slave side is not currently supported by the upstream qmk firmware. However, it is possible to implement a custom transport which allows it to use it on the slave side. I'd recommend to ask drashna about it, who did this already. 
I hope that resolves your question!

hyphonetics, 2021-08-10:
Awesome, that helps a lot. "Not currently supported" makes me feel better about it not working, I'm glad it wasn't just something I was missing. I may just make the right side my master and change the matrix/keymap in the firmware. 
Appreciate the help, thanks for making such an awesome looking little split board as well!

MangoIV, 2021-08-10:
Glad you like it, enjoy your board!

closed 2021-08-10 by MangoIV