MangoIV/dracuLad #7: Clearer documentation, maybe with pictures?

jpabst, 2021-09-22:
This is my first keyboard build, and maybe that's the problem... but I'm familiar with soldering etc..

the pinout isn't clear, so I didn't know the right orientation to put in my Elite-C.  Thankfully there's a picture for one side which helped (your not-elite-c that you feature isn't helpful since it doesn't show the pinout on it at all)

Your instructions state to solder in the diodes on the bottom, but it isn't clear where to solder the leds?  Is that also the bottom?
You have pictures: acryli_no_caps_top.jpeg that show the diodes soldered to the top?!?!?

I didn't know to be careful about aligning the pins inside the square indicator for the controller, so I've got them on the wrong pins for one of my PCB's (so I'll unsolder it and move it, but it'll be painful).

The main issue I have right now:  my LEDS won't turn on.  I put them on the top, but I'm guessing that's wrong.  I tried putting one on the bottom of the other board... but while it flashed green briefly (when pressed to the contacts), after I soldered it in... it won't turn on at all (note, that's the only one I had put in.

Also, you have pictures of the OLED jumpers filled in without the OLED being installed (acryli_no_caps_top.jpeg), also have pictures of the trackball jumpers filled in... but no trackball.  So I'm not sure whether I'm supposed to fill them in or not (I don't have the trackball... yet)

MangoIV, 2021-09-22:
Hi!
First off, I want to tell you that this is a more advanced project, targeting mainly people who already built a keyboard and/or have a little bit of experience with that kind of hardware. Many of your questions could be answered looking at the kicad schematic, for example or by common sense. However, I admit that the documentation is a bit sparse and I could do something about that. Now to your questions:
- The orientation of the controllers is always with the MCU facing towards the PCB if you put your pinheaders in the outlined spots on the PCB, you can also match the obvious GND pins (they have little crosses around them) with the GND pins on the controller, if you're unsure
- On the picture the diodes might be soldered on top, this is absolutely possible, as you can clearly see, the under- and topside are connected by Vias within the pads. Soldering them on the bottom is recommended for the simple and practical reason that it is easy to reach, even after soldering in the switches. 
- it doesn't really matter whether you put the LEDs on the top or the bottom, they should work in any way. However, they are obviously intended to be soldered on the bottom side, they're for underglow, after all. If you face issues with the LEDs, check them one by one, they are daisy chained. The first LED in the chain should be the one directly under the controller and the other ones just follow in a line. It is quite probable you killed them, they're made of plastic and very sensitive to heat. As a tip, never touch the plastic but always touch only the pins/pad. 
- The jumpers can be soldered, even if you don't plan to use an OLED or trackball at that time. It's crucial to solder them on the side the ball/OLED is going on, though, the jumpers basically serve as a way to make the PCB reversible by reversing the pinout corresponding to the jumpers you solder 

On a side note: This is an open source project and only a hobby of mine. I am neither responsible for providing a perfect buildguide (although that would be optimal) and neither for other peoples builds. I try to provide as much support as possible but it's not an obligation of mine to do so.

I hope you get your build to work and I'd be happy about a contribution to the buildguide!

Kind regards.

jpabst, 2021-09-23:
Sorry if I came across as demanding, it was late and I was a little frustrated.  I actually went through the different project folders and while it may have given me the pinout in the kicad schematics, I have no idea what kicad is...  I was thinking this morning that I should just put in a pull request with what would have helped me.  the LED's I put on the bottom of the right-hand side work great, it's just the ones on the left-hand (that I did first) that are all probably broken, since desoldering them is tricky and I probably broke them in the process.  Still, the keyboard works!

Thanks for this honestly.  I was looking into the Kyria but supply was a problem for me and I stumbled across this one that had the similar aggressive pinky offset and dove in.

Thanks for the answers and your time

-Jon

closed 2021-09-23 by jpabst

MangoIV, 2021-09-23:
Hi,
Just opening that up for a second again to remind you that the LEDs might not all be broken but just a few. Remember, they are daisy chained...

I wish you good luck and great fun with your build and hope you'll get everything working soon!