MangoIV/dracuLad #9: Can't get encoders to work

stefanwatt, 2022-01-21:
I'm using the nice!nano with ZMK on my dracuLad.
Everythings working fine except for the encoders. The keypress works but the rotary action does not.
I followed the [ZMK docs](https://zmk.dev/docs/development/new-shield#encoders) for enabling encoder support.

MangoIV, 2022-01-21:
Hi!

I'm sorry to tell you, that I did not have time to implement ZMK support yet, so I cannot give you working code. 
However, you should be aware that generally, ZMK only supports encoder on the central side, not on the peripheral side. 

If you think, this is a firmware issue, please be so kind and visit their discord server, for which you can find a link on their website. 

If you think, this is a hardware issue, please be aware that all of the encoders are confirmed working on QMK so it's probably a cold joint or a broken encoder. 

In case of the latter, you can reach me on the [40% discord](https://discord.gg/zjZjvNkhpW) or on the [German mechanical keyboard discord](discord.gg/mkd). 

Good luck with building the DracuLad, I hope I can help you on the discords or you can find help with the zmk guys :)

closed 2022-01-28 by MangoIV

brandon-page, 2025-02-24:
@stefanwatt Did you get this working?

stefanwatt, 2025-02-24:
Yes there is a ZMK fork that allows you to use both encoders separately. I'm sure you can find it