MangoIV/dracuLad #10: It's crucial to solder trackball/OLED jumpers even if you don't plan to them?

lpanebr, 2022-03-21:
Hi and thank you sooo much for this great layout!

This is my first built and although I had no experience in soldering anything other than two wires together, I'm careful and like to study in preparation for challenges.

I will not add an OLED or trackball now and will use all the 36 switches as keys, so no rotary encoders also.

I've read [your build guide](https://github.com/MangoIV/dracuLad/blob/master/buildguide.md#draculad-build-guide) and every other guide/video I could find on DracuLad layout (very few unfortunately). I was about to just ignore the jumpers until I read your comment on #7:

> The jumpers can be soldered, even if you don't plan to use an OLED or trackball at that time. It's crucial to solder them on the side the ball/OLED is going on, though, the jumpers basically serve as a way to make the PCB reversible by reversing the pinout corresponding to the jumpers you solder.

Then I thought I'd better ask: *What do you mean by "The jumpers can be soldered, even if you don't plan to use..."?*

Well... while reading your build guide I noticed the referenced crkbd link for socketing the controller is broken. I searched and found this https://github.com/foostan/crkbd/blob/main/corne-classic/doc/buildguide_en.md which I believe is it(?).

Reading [this crkbd buid guide](https://github.com/foostan/crkbd/blob/main/corne-classic/doc/buildguide_en.md) I've found the socketing controller information and also the information below (what you know!?).

> To use OLED modules, short circuit the jumper patterns. Only short circuit the front side

Reading this I did not fully understood if I should 1) short circuit as to close each individually, or 2) short circuit them all with each other?

I then inspected the pictures and am pretty positive it's 1: *Short circuit jumpers as to close each one individually.*
[image]
[image]

*Just for some peace of mind could you please confirm:*
 - if it's this what you meant by "it's crucial to solder them..."?
 - in your layout this should be done to each jumper set (OLED and Trackball)?
 - is it possible to add an OLED later on?

Thanks again!!!

PS: I've attached some pictures just because I'm very happy! I've only done one hand diodes and LEDs yet.

[image]

[image]

MangoIV, 2022-03-21:
Generally, if you want to add an OLED, you have to solder the jumpers **on the top side** that say "oled jumpers". You do it just like that for the pimoroni jumpers. 

If you don't plan to use the trackball nor oled, you can just leave them unsoldered but it doesn't hurt to solder them if you decide otherwise in the future. I hope this clarifies it. 

Nice soldering, looks good! Have fun with your board!

lpanebr, 2022-04-22:
Hi. I've finished but there only there's something weird:
Both sides are flashed. When the left side is plugged in through USB, all switches work but the right side doesn't work at all (except for the Pro Micro green LED that stays on). If I plug the USB into the right side, the switches work as the left side but mirrored.

I've checked continuity from corresponding TRSS jack connections and from them to their corresponding Pro Micro pin and all is good.

The problem is the same with the OLED jumpers connected or not.

I did not try connecting the trackball as you said them don't need to be jumped.

**Any idea of what I could try?**

MangoIV, 2022-04-23:
Have you flashed the eeprom to remember the correct side? If not you can flash a different firmware on each side , one for left and one for right.

lpanebr, 2022-04-23:
Ow no. I've flashed the same hex I compiled from the draculad keyboard and keymap files on the qmk repo!
Are there different keyboard or keymap files?

lpanebr, 2022-04-23:
Hmmm.. I found this and am trying to understand it:
https://docs.qmk.fm/#/feature_split_keyboard?id=handedness-by-eeprom

lpanebr, 2022-04-23:
I had to change `#define EE_HANDS` with `#define SPLIT_USB_DETECT`, recompile and reflash both sides. Now it works!

MangoIV, 2022-04-23:
Awesome. The article you linked was correct and contains all the information you need if you want handedness by eeprom to work ;)

lpanebr, 2022-04-23:
When I finish I'll send a PR to include my keymap to the draculad keyboard layout.

MangoIV, 2022-04-24:
I wouldn't do that. Qmk already includes enough keymaps and the PR will last an eternity to merge. Please don't keep the maintainers busy with things like that, except if your keymap contains something really innovative.

lpanebr, 2022-04-25:
OK.

MangoIV, 2022-04-27:
Do you have any other issues? If not I'd close this ;)

lpanebr, 2022-04-27:
Actually on the left board only two LEDs are working.
[image]

But I think the cause is probably that I have ruined this one:
[image]

so I'll just replace it.

Yes, you may close. Thanks for this awesome PCB and for all your support!

closed 2022-04-27 by lpanebr