MangoIV/dracuLad #16: PCB Pinout

brandon-page, 2024-05-12:
Apologies if there is an obvious file with this info but I didn't see it.  Does anyone have a PCB Pinout diagram for the controller? 

I would like to use a Seeed Studio XIAO NRF52840, which is shorter than a Pro Micro but a 3x5+3 matrix doesn't take many pins. Since I am going wireless/ZMK I don't care about LEDs or trackball and could live without encoders if I had to. If some of the matrix pins are lower down (where they would be missing on my controller) I could wire jumpers, I would just need to know which pins.  

Thanks!

MangoIV, 2024-05-12:
The kicad files are available. They also have the pinout. I don’t think the xiao will be a possibility if I had to guess. Not enough pins.

closed 2024-05-12 by brandon-page

brandon-page, 2024-05-12:
[image]